SignalK/vedirect-serial-usb #29: Support more than one (1) USB-ve.direct dongle

LeaseOnLife, 2021-01-30:
I have 4, with more coming... One for each solar controller and one for each of two battery monitors. So far I am using EmonCMS and EmonHub to record/log/graph data, and maybe able to help with mapping from BMV-700, BMV-712, MPPT smartsolar and blue-solar devices.

Thanks
Dirk

Leijoma, 2021-03-29:
This is something i also would appreciate. I have the Victron Smartshunt as well as a solar charge controller.

joabakk, 2021-04-09:
This request is gaining popularity. I would be happy for a Pull Request tested in real life

Leijoma, 2021-04-09:
I have a Solar charger and Smart shunt that i have hooked up at home if you want help with testing.

KEGustafsson, 2021-04-09:
I have been modifying current version to support multiple devices (serial, udp, tcp). There are still couple of things to do before PR, but I have now already almost working setup. I hope I can finalize mods during weekend and then I can do a PR.

Leijoma, 2021-04-10:
I tested this branch. I was able to create two devices,  but got the following error message in the plugin config: Status: Failed to start: Cannot read property '0' of null.

The config file looks like this after creating two devices:

{
  "configuration": {
    "vedirect": [
      {
        "device": "Serial",
        "connection": "/dev/ttyUSB0",
        "port": 7878,
        "ignoreChecksum": true,
        "mainBatt": "House",
        "auxBatt": "Starter",
        "solar": "Main"
      },
      {
        "device": "Serial",
        "connection": "/dev/ttyS0",
        "port": 7878,
        "ignoreChecksum": true,
        "mainBatt": "House",
        "auxBatt": "Starter",
        "solar": "Main"
      }
    ]
  },
  "enabled": true
}

KEGustafsson, 2021-04-11:
@Leijoma I pushed a fix to that issues, so it should be working now. Can you test again?
Give each device mainBatt, auxBatt and solar unique names. These names are used for SK paths.

Leijoma, 2021-04-11:
I have downloaded the fix and now i get data from both devices! However it looks like all data (from both devices) is displayed in each device. See screenshot. 

[image]

KEGustafsson, 2021-04-11:
@Leijoma Now you pointed this out, I can see the same issue. I will start to investigate what is wrong and how to correct that.

KEGustafsson, 2021-04-11:
@Leijoma Now the parser issue should be corrected. Can you test once?
On my trials, this is working OK with tcp and udp connections.

Leijoma, 2021-04-11:
@KEGustafsson Looks great. 

My setup is:
 - Raspberrry Pi 4
 - Smart shunt connected to the serial port (with a level converter)
 - Solar charger connected  through a Victron VEDirect USB cable.

Output looks like this:
electrical.batteries.House_solar.current           | 0.          | A | 04/11 16:15:46 | vedirect-signalk.XX
electrical.batteries.House_solar.voltage.          | 11.49    | V | 04/11 16:15:46 | vedirect-signalk.XX
electrical.batteries.Starter_shunt.voltage         | -0.014 | V | 04/11 16:16:34 | vedirect-signalk.XX
electrical.charger.House_solar.chargingMode | "off"    |     | 04/11 16:16:35 | vedirect-signalk.XX

Thank's for fixing this!

/Magnus

closed 2021-04-22 by joabakk