SignalK/vedirect-serial-usb #37: Power Charging

spidgrou, 2021-06-10:
Hi,
is it possible to see:

- current or power that is charging the battery (at the moment is only available the panelPower)

thanks
Max

joabakk, 2021-06-10:
Hi,

This plugin will translate vedirect to signal K. It does not add any other information. 

Is there a specific variable you are looking for, that is available somewhere else?

spidgrou, 2021-06-11:
Hi,
as I need is one information:
- BATTERY CURRENT, how much current the mppt regulator (in this case) send to the battery

This is a different information from Panel Power, that is how much power the panel is generating

Is it possible to have?

Thanks
Max

joabakk, 2021-06-29:
If you have a code for it, we should be able to add it. If you look at https://github.com/SignalK/vedirect-serial-usb/blob/master/lib/fields.js, there are codes for each item this plugin reports. I would have thought that 'electrical.solar.*.loadCurrent' covered your needs?

closed 2021-06-29 by joabakk