SignalK/vedirect-serial-usb #38: Charging mode - CS

spidgrou, 2021-06-11:
Hi,
I'm using two different MPPT controller bluesolar 75/15
At some point of the day, one of the two is still in 'Absorption' (CS=4) but the other is still in 'Bulk' (CS=3).
I see that there is a state of charge signalK "electrical.charger.House.chargingMode" but is it refererred to which of the two MPPT Regulator connected?
I think on both, because if I switched off one of the two MPTT controller, the value jump from 'bulk' to 'off' continuously.
Is it possible to move this information (State fo Charge) sentence under every single MPTT controller? so we can monitor how the MPPT controller works?

Many thanks.
Max

KEGustafsson, 2021-06-15:
@spidgrou Are you using same SK path, in this case "House", multiple times?
If yes, then change one device. e.g. to "House2".

If same path is used then two or more devices write to same SK path and you might see this kind of issues.

spidgrou, 2021-06-19:
@KEGustafsson thanks for your help.
Your solution is a good workaround, I can confirm that it works. 
I hope to see in one of the next update the CS information under the right tree. 
Many thanks.

KEGustafsson, 2021-06-19:
@spidgrou This is not the workaround, but the way the pluging should be used. It is stated at Readme to use separate SK paths for each devices.

Or if you mean something else approach, please can you explain it?
Improvements are always welcome.

spidgrou, 2021-06-19:
@KEGustafsson I try to explain my point of view.
In the Plugin setup we insert data for House, Aux and Panel Name.
It is correct because the array I can connect to one of them (not so common to connect to Aux/engine battery, but it could); so the device send the battery voltage and the battery current to the 'HOUSE' path.
Normally boat and RV have only one House bank battery.

All the other details from MPPT, all of them with exception of the two over here, are informations related to the array/MPPT device.

In my opinion, all of them need to be under the same signalK path so we can view all the details in the same path.
The CS parameter (state of operation), it is something that is a state that is related to the MPPT or to the charger and not to the bank battery, because for example I can have some MPTT in 'bulk' and others in 'off' and the charger in float.

I hope to be enough good to explain my thought, English is not my native language so please be patient.
Ciao
Max

KEGustafsson, 2021-06-20:
@spidgrou Then you should include device reference also to this field. E.g mppt1-House, mppt2-House, source-destination, etc...

Each device need to have independent SK naming to avoid collision.

closed 2021-06-29 by joabakk