SignalK/vedirect-serial-usb #39: Battery current (Label I) not relayed from MPPT charger to Signal K

Leijoma, 2021-07-13:
My Victron MPPT 75/15 is transmitting Battery current (Label I) over ve.Direct, but it does not show up in signal K. Is it a bug, a missing feature or something i am doing wrong?

I also have a smart shunt connected so i am working with two devices. 
All other info look OK.

/Magnus

Leijoma, 2021-07-13:
Took a look at the code and it looks like the I label will translate to batteries.house.current for both devices. I'm not an expert coder so i don't think i can help, but it would be great if it was possible to separate the measurments. Maybe by adding a custom field to the path in the settings for each vedirect device.

KEGustafsson, 2021-07-13:
@Leijoma Do you use "house" naming with your both device? Otherwise it should not be possible to write same SK path.
Can you point to the code location that you expect to have an error?

Leijoma, 2021-07-13:
Ah
Now I understand. I used the same name because they both measure the house
bank...

No problem, my mistake.

/Magnus

closed 2021-07-17 by joabakk