SignalK/vedirect-serial-usb #46: Can't find "electrical.batteries.House.capacity.dischargeSinceFull" in bmv-712

vgoya, 2023-01-21:
Trying to read Ah you recommended this path: "electrical.batteries.House.capacity.dischargeSinceFull" but I can't find it from the BMV-712

Only data in C I can read is "electrical.batteries.House.lifetimeDischarge"

I can also read: "electrical.batteries.House.capacity.consumedCharge" with no units

Thank you in advance

V

xbgmsharp, 2023-05-07:
The only value that don't get parsed is `MON`. is that dischargeSinceFull?

I don't get any C data.

consumedCharge the unit is present but probably off?
https://github.com/SignalK/vedirect-serial-usb/blob/master/lib/fields.js#L203

BMV-712 Smart Output
```
PID	0xA381
V	13720
VS	13357
I	0
P	0
CE	0
SOC	1000
TTG	-1
Alarm	OFF
Relay	OFF
AR	0
BMV	712 Smart
FW	0413
MON	0
Checksum	`
H1	-1252
H2	0
H3	0
H4	0
H5	0
H6	-3155
H7	28
H8	14449
H9	0
H10	0
H11	0
H12	0
H15	22
H16	14453
H17	3
H18	3
```

xbgmsharp, 2023-08-29:
Now i also get the no unit for this electrical.batteries.House.capacity.consumedCharge`
But it seem correctly declared ,https://github.com/SignalK/vedirect-serial-usb/blob/master/lib/fields.js#L203