SignalK/vedirect-serial-usb #60: SmartSolar MPPT Current

tonipezzoni, 2023-12-30:
Hello,
The screenshot is showing all the paths received from a SmartSolar MPPT via ve.direct USB with SK 2.41 and plugin 1.5.0.
The current form panels is "electrical.solar.Panels.loadCurrent", and the current to the batteries is "electrical.batteries.Services.current" right?

Well, the electrical.solar.Panels.loadCurrent is always zero.
Any idea?
Thank you!
[image]

tonipezzoni, 2023-12-30:
Probaly the path "electrical.solar.Panels.loadCurrent" is referred to the optional load, not connected.
If so, why the current from panels don't have a path?
As electrical.solar.Panels.panelVoltage and  electrical.solar.Panels.panelPower, a  electrical.solar.Panels.panelCurrent should be present.
Thank you

KEGustafsson, 2024-01-01:
Your output `electrical.solar.Panels.load` is `off`. If I don't remeber totally wrong, this is output switch indicator for `electrical.solar.Panels.loadCurrent`. So if it is not `on` then current will remain 0.

For some reason, there is no panel current available from MPPT. If you need that, it must be calculated separately, `I=P/U`